chu11/freeipmi-mirror #4: add a no-authentication option

zenfish, 2016-07-23:
The Get DCMI Capabilities Info command along with the IPMI commands get authentication, get system guid, get channel auth capabilities, get cipher suites, and get pet ack (!) require no authentication or session at all, according to the IPMI specification, but there is no way to remotely execute these commands with FreeIPMI tools without having to supply a valid username and password to the BMC.

It'd be great to have a --no-auth/--no-session or some flag that would allow you to execute these without authentication.

Thanks!

chu11, 2016-07-25:
So I see two ways to do this.

1) provide a --no-auth/--no-session option to ipmi-raw.  So users could send a single payload this way.  The option doesn't quite make sense with any other tool (e.g. ipmi-sensors) b/c the tools are so tied to doing a sequence of packets to accomplish a purpose (e.g. get all the sensors).

2) provide options to ipmi-ping to specify alternate packets to use besides get authentication capabilities.  But I think it would be only useful/make sense with a handful of packets (e.g. get channel auth, get system guid, etc.).  Doesn't really make much sense with some other packets.

chu11, 2016-07-27:
I had completely forgotten about this, but a long time ago I added

````
 * NOSESSION - for outofband only, do not create an IPMI session.
 * Useful for the few IPMI payloads that do not require a session for
 * an IPMI command to be sent (e.g. Get Channel Authentication
 * Capabilities, Get System GUID, PET Acknowledge).  Can only be set
 * during opening, not later using ipmi_ctx_set_flags().  If set, you
 * cannot call most IPMI payload functions, only those few that send
 * data without a session.
````

into the core libfreeipmi API.  So adding this as an option to ipmi-raw seems simple enough.  Would this suit your needs?

chu11, 2016-07-28:
I went ahead and did this and put it in the branch "ipmirawnosession".  Could you try it out and LMK.  I verified that "get auth capabilities" and "get session challenge" worked outside of a session.  It appears my motherboard doesn't support "get system guid" outside of a session though.

zenfish, 2016-07-28:
Dude!

You’ve forgotten more about IPMI than I’ll ever know!

I’ll check it out ;)

dan

zenfish, 2016-07-28:
Yep, get system guid only works on some… I can’t remember the #’s… but I seem to recall something like over half, but not all.

A curious one because it gives you a unique ID for a chassis, which is otherwise pretty impossible to get (putting on the security scanning hat….)

dan

chu11, 2016-08-09:
Hey dan, just wondering how it's working out.

zenfish, 2016-08-09:
I haven’t forgotten!

I tried it on an old 32 bit machine -

	ipmi-dcmi: error while loading shared libraries: libfreeipmi.so.17: cannot open shared object file: No such file or directory

I thought oh, that’s just some old machine, but 1.4.3 did work on it (just retested.)

Then I tried it on a 64 bit VMware image and had the same issue. 

Spurred on by your note… these are both ubuntu 14… so I hunted around and tried it with this option on ./configure (on the 64 bit) -

	--libdir=/lib64

Which created a usable binary.

I’ll try to test this in the next day or so, sorry for the delay!

dan

chu11, 2016-08-09:
hmmmm, was 1.4.3 a build from scratch or installed via a distro?  I'm a little surprised that /lib64 works, but every distro and setup is a little different.

zenfish, 2016-08-09:
Build from scratch; that just happened to be the last time I built it from the 
last ‘round of things I was looking at, nothing special about it in particular.

But yes, every snowflake and all that… stupid distros, lol. Mostly just an FYI.

dan

chu11, 2016-09-27:
I've merged this into the master since it works for me and seems generically useful.  LMK if there's any issues that pop up and we can fix from there.

chu11, 2017-01-23:
closing as its merged into the mainline now

closed 2017-01-23 by chu11