chu11/freeipmi-mirror #8: ipmi-config ipv6 support

chu11, 2016-11-23:
per discussion in #6 , split off from that ticket

chu11, 2017-01-09:
Here's some initial thoughts on the more manual activities that have to be done to get IPv6 support into ipmi-config.

libfreeipmi/include/freeipmi/spec/ipmi-lan-configuration-parameters-spec.h
- [ ] - Table 23-4 new IPv6 configs

libfreeipmi/include/freeipmi/cmds/ipmi-lan-cmds.h
libfreeipmi/cmds/ipmi-lan-cmds.c
- [ ] - Table 23-4 new templates
- [ ] - Table 23-4 new fill functions
- [ ] - new macros needed for config parameters/options?

libfreeipmi/include/freeipmi/templates/ipmi-lan-cmds-templates.h
- [ ] - document new templates

libfreeipmi/include/freeipmi/api/ipmi-lan-cmds-api.h
libfreeipmi/api/ipmi-lan-cmds-api.c
- [ ] - Table 23-4 new API equivalents

23.2a DHCPv6 Timing Parameters
- [ ] - macros/enums needed?

23.2b Neighbor Discovery / SLAAC Timing Parameters
- [ ] - macros/enums needed?

lamontj, 2017-01-09:
Presumably this work wants to be based on the branch from #6.  Also note that we'll need to allow both IPv4 and IPv6 addresses to be configured simultaneously.  (Though my immediate need is discovering the already-assigned-by-the-admin IPv6 address on a machine with no IPv4 addresses at all.)

chu11, 2017-01-09:
Ya know, these were my more base thoughts on just the manual activities to add basic protocol support in, which is sort of separate from the ipmi-config issue.  Let me fork these manual activities off into a separate issue.

lamontj, 2017-01-13:
Just to get the discussion going on this point:
I'm guessing that this will become something along the lines of Lan_Conf:IPv6_Address or such?  (or would that be plural?)  separate out dynamic and static, or leave them together?

chu11, 2017-01-13:
I haven't thought too far ahead about it.  B/c the IPv6 config is so different and somewhat separate of IPv4, I think it might be best to have an entire section like "Lan_Conf_IPv6" like:

Lan_Conf_IPv6:IP_Address_Source
Lan_Conf_IPv6:IP_Address
Lan_Conf_IPv6:IP_Address_Router
Lan_Conf_IPv6:MAC_Address_Router

etc.

If we have them all in the "Lan_Conf" section, I just think it'll get complicated for both users and for developers trying to wrap their heads around it if both IPv4 and IPv6 are supported?  That's just my initial feeling.

Naturally, if only IPv6 is supported and no IPv4 is supported, the appropriate fields in LAN_Conf will not be checked out for the user.  If only IPv4 is supported, the entire "Lan_Conf_IPv6" section will not checked out for the user.

Of course what I say above is sort of long term.  For initial trials, feel free to do "Lan_Conf:IPv6_Address" and "Lan_Conf:IPv6_Static_Enable" or whatever.  I imagine there will be some "figuring things out" as we go along.

chu11, 2017-08-23:
completed in master branch, mostly for static config only.  Can add dynamic later.  Considering this complete now.

closed 2017-08-23 by chu11