chu11/freeipmi-mirror #11: add new IPMI errata #6 IPv6 support into libfreeipmi

chu11, 2017-01-09:
This is a fork off of issue #8   

Here's some initial thoughts on the more manual activities that have to be done to get IPv6 support into ipmi-config.  Some times unknown w/ quesiton marks attached

libfreeipmi/include/freeipmi/spec/ipmi-lan-configuration-parameters-spec.h
- [ ] - Table 23-4 new IPv6 configs

libfreeipmi/include/freeipmi/cmds/ipmi-lan-cmds.h
libfreeipmi/cmds/ipmi-lan-cmds.c
- [ ] - Table 23-4 new templates
- [ ] - Table 23-4 new fill functions
- [ ] - new macros needed for config parameters/options?

libfreeipmi/include/freeipmi/templates/ipmi-lan-cmds-templates.h
- [ ] - document new templates

libfreeipmi/include/freeipmi/api/ipmi-lan-cmds-api.h
libfreeipmi/api/ipmi-lan-cmds-api.c
- [ ] - Table 23-4 new API equivalents

23.2a DHCPv6 Timing Parameters
- [ ] - macros/enums needed?

23.2b Neighbor Discovery / SLAAC Timing Parameters
- [ ] - macros/enums needed?

lamontj, 2017-01-11:
My ipv6 branch has (untested) configs and fill functions for your review -- no new templates yet, though.

chu11, 2017-01-11:
@lamontj Could you just go ahead and send a pull request.  It's easier to review/comment via the PR review system.

lamontj, 2017-01-11:
@chu11 Before I write fill functions, can I get feedback on the configs and templates?  After that I may go for full-depth on a few of the config variables, so that I can get fully familiar with the path earlier.

chu11, 2017-01-12:
@lamontj that's my plan, will review hopefully starting tonight

chu11, 2017-08-01:
Hey @lamontj, just looking over things, and noticed the "commit" parts of the IPv6 in ipmi-config were never completed.  Is this still on your todo?

chu11, 2017-08-23:
majority of this is complete, discounting a few DHCP/dynamic IPv6 stuff.  Will add if needed.  Considering this done.

closed 2017-08-23 by chu11