chu11/freeipmi-mirror #22: Toggle executable name in syslog

sevagh, 2018-02-26:
Hello,

ipmiseld is putting stuff in syslog with the full executable path:

```
Feb 26 11:44:11 app-server-1 /usr/sbin/ipmiseld[30021]: SEL System Event: Feb-26-2018, 19:39:24, Sensor #0, Warning, Correctable memory error
```

It would be cool if it just logged with the name `ipmiseld` (stripping `/usr/sbin`) - is this toggleable from the config options?

chu11, 2018-02-26:
It appears I simply pass ```argv[0]``` to ```openlog()```.

```
openlog (argv[0], LOG_ODELAY | LOG_PID, prog_data.log_facility);
```

So if you don't run the full path, that would solve the problem temporarily.

But you are correct, I should strip off all the preceding path stuff before passing the daemon name to ```openlog()```.  I'll add that to my todo to cleanup.

sevagh, 2018-03-08:
I can make a PR for this. Should I just hardcode `ipmiseld` (my preference), or `argv[0]` with path stripping?

closed 2018-03-20 by chu11