chu11/freeipmi-mirror #35: receiving "SEL illegal timestamp situation" in logs when using ipmiseld

adamwoolhether, 2020-09-03:
As the title says, my rsyslog files are populated only with a single entry for every SEL event: "SEL illegal timestamp situation". 

This happens with default or custom-defined string output.

I can generate accurate reports with the '--test-run' option.

here's an example of my output
```
ipmiseld -v --poll-interval=5 --log-facility=LOG_LOCAL4 --system-event-format '%i %d %t %T %s %I %E' --oem-timestamped-event-format '%i %d %t %I %o'
```

and my rsyslog config in /etc/rsyslog.d/ipmiseld.conf:
```
# BMC System Event Log events
local4.*                                                /var/log/ipmiseld.log
```

Output for all events looks like:
```
Sep  3 10:20:44 xxx ipmiseld[359204]: SEL illegal timestamp situation
Sep  3 10:25:45 xxx ipmiseld[359204]: SEL illegal timestamp situation
Sep  3 10:40:22 xxx ipmiseld[8106]: SEL illegal timestamp situation
```

chu11, 2020-09-19:
sorry for late reply, completely missed this.

What board are you running on and could you should the output of `ipmi-sel -i`.  The "SEL illegal timestamp situation" indicates that the timestamps & record counts listed by the SEL are out of whack.

```
          /* This shouldn't be possible and is likely a bug in the
           * IPMI firmware (user erased entries but timestamp didn't
           * update, SEL added entries and updated timestamp but
           * didn't update entry count, etc.) we'll only save off the
           * host state for later.
           */
          if (host_data->prog_data->args->verbose_count)
            ipmiseld_syslog_host (host_data, "SEL illegal timestamp situation");
```

adamwoolhether, 2020-09-21:
Hi @chu11 , no worries, thanks for the response.

I've ran the command on a series of Quanta servers, all of which experience the problem:
`for i in 1 2 3 4 5 6 7; do ssh 10.108.5.$i 'dmidecode -s system-product-name' && ipmi-sel -i -h 10.108.2.$i -u XXX -p XXX ; done`

```
QuantaGrid Q72D-2U
SEL version                            : 1.5
Number of log entries                  : 398
Free space remaining                   : 9198 bytes
Recent addition timestamp              : 08/14/2020 - 06:42:34
Recent erase timestamp                 : Post-Init 0 s
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 511
Largest free block                     : 511 allocation units
Maximum record size                    : 1 allocation units

QuantaGrid D52G-4U
SEL version                            : 1.5
Number of log entries                  : 286
Free space remaining                   : 11214 bytes
Recent addition timestamp              : 09/16/2020 - 00:53:05
Recent erase timestamp                 : 05/14/2020 - 00:14:47
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 623
Largest free block                     : 623 allocation units
Maximum record size                    : 1 allocation units

QuantaGrid D52BV-2U
SEL version                            : 1.5
Number of log entries                  : 80
Free space remaining                   : 14922 bytes
Recent addition timestamp              : 09/10/2020 - 06:47:03
Recent erase timestamp                 : Post-Init 0 s
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 829
Largest free block                     : 829 allocation units
Maximum record size                    : 1 allocation units

QuantaGrid D52B-1U
SEL version                            : 1.5
Number of log entries                  : 233
Free space remaining                   : 12168 bytes
Recent addition timestamp              : 09/02/2020 - 02:04:15
Recent erase timestamp                 : 03/05/2020 - 05:11:52
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
[... 51 lines cut ...]
Number of free allocation units        : 0
Largest free block                     : 0 allocation units
Maximum record size                    : 1 allocation units
```

Running a standard ipmi-sel also results in the following error:
```
QuantaGrid Q72D-2U
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaGrid D52G-4U
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaGrid D52BV-2U 28S5BMAXXXX
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaGrid D52B-1U
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaPlex T42D-2U
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaPlex T42S-2U(LBG-4) -
ipmi_sdr_parse_sensor_units: invalid sdr record
QuantaGrid D43K-1U
ipmi_sdr_parse_sensor_units: invalid sdr record
```
Adding the `--legacy-output` tag *does* work, but output can't give the 'priority-level'.

Interestingly, I *can* run the ipmi-seld command with the `--test-run` flag successfully with accurate output, but daemon cannot.

chu11, 2020-09-21:
lets start with ipmi-sel, b/c it appears that something on your board is non-standard and freeipmi can't seem to handle it.

Could you do `ipmi-sel --debug`. If it's a lot of records, could you limit it by just outputting the first record w/ `--display=RECORDID`.  Thanks.

adamwoolhether, 2020-09-22:
Hi Chu, here's my out output from one of the D52G. I'm not able to isolate based on RecordID in the given output.
```
$ ipmi-sel -h 10.108.2.91 -u xxx -p xxxx --debug

10.108.2.91: =====================================================
10.108.2.91: IPMI 1.5 Get Channel Authentication Capabilities Request
10.108.2.91: =====================================================
10.108.2.91: RMCP Header:
10.108.2.91: ------------
10.108.2.91: [               6h] = version[ 8b]
10.108.2.91: [               0h] = reserved[ 8b]
10.108.2.91: [              FFh] = sequence_number[ 8b]
10.108.2.91: [               7h] = message_class.class[ 5b]
10.108.2.91: [               0h] = message_class.reserved[ 2b]
10.108.2.91: [               0h] = message_class.ack[ 1b]
10.108.2.91: IPMI Session Header:
10.108.2.91: --------------------
10.108.2.91: [               0h] = authentication_type[ 8b]
10.108.2.91: [               0h] = session_sequence_number[32b]
10.108.2.91: [               0h] = session_id[32b]
10.108.2.91: [               9h] = ipmi_msg_len[ 8b]
10.108.2.91: IPMI Message Header:
10.108.2.91: --------------------
10.108.2.91: [              20h] = rs_addr[ 8b]
10.108.2.91: [               0h] = rs_lun[ 2b]
10.108.2.91: [               6h] = net_fn[ 6b]
10.108.2.91: [              C8h] = checksum1[ 8b]
10.108.2.91: [              81h] = rq_addr[ 8b]
10.108.2.91: [               0h] = rq_lun[ 2b]
10.108.2.91: [              3Ah] = rq_seq[ 6b]
10.108.2.91: IPMI Command Data:
10.108.2.91: ------------------
10.108.2.91: [              38h] = cmd[ 8b]
10.108.2.91: [               Eh] = channel_number[ 4b]
10.108.2.91: [               0h] = reserved1[ 3b]
10.108.2.91: [               0h] = get_ipmi_v2.0_extended_data[ 1b]
10.108.2.91: [               3h] = maximum_privilege_level[ 4b]
10.108.2.91: [               0h] = reserved2[ 4b]
10.108.2.91: IPMI Trailer:
10.108.2.91: --------------
10.108.2.91: [              4Eh] = checksum2[ 8b]
10.108.2.91: =====================================================
10.108.2.91: IPMI 1.5 Get Channel Authentication Capabilities Response
10.108.2.91: =====================================================
10.108.2.91: RMCP Header:
10.108.2.91: ------------
10.108.2.91: [               6h] = version[ 8b]
10.108.2.91: [               0h] = reserved[ 8b]
10.108.2.91: [              FFh] = sequence_number[ 8b]
10.108.2.91: [               7h] = message_class.class[ 5b]
10.108.2.91: [               0h] = message_class.reserved[ 2b]
10.108.2.91: [               0h] = message_class.ack[ 1b]
10.108.2.91: IPMI Session Header:
10.108.2.91: --------------------
10.108.2.91: [               0h] = authentication_type[ 8b]
10.108.2.91: [               0h] = session_sequence_number[32b]
10.108.2.91: [               0h] = session_id[32b]
10.108.2.91: [              10h] = ipmi_msg_len[ 8b]
10.108.2.91: IPMI Message Header:
10.108.2.91: --------------------
10.108.2.91: [              81h] = rq_addr[ 8b]
10.108.2.91: [               0h] = rq_lun[ 2b]
10.108.2.91: [               7h] = net_fn[ 6b]
10.108.2.91: [              63h] = checksum1[ 8b]
10.108.2.91: [              20h] = rs_addr[ 8b]
10.108.2.91: [               0h] = rs_lun[ 2b]
10.108.2.91: [              3Ah] = rq_seq[ 6b]
10.108.2.91: IPMI Command Data:
10.108.2.91: ------------------
10.108.2.91: [              38h] = cmd[ 8b]
[... 1139 lines cut ...]
10.108.2.91: [               0h] = message_class.reserved[ 2b]
10.108.2.91: [               0h] = message_class.ack[ 1b]
10.108.2.91: IPMI Session Header:
10.108.2.91: --------------------
10.108.2.91: [               0h] = authentication_type[ 8b]
10.108.2.91: [        24BDFC5Fh] = session_sequence_number[32b]
10.108.2.91: [        A58CCC5Fh] = session_id[32b]
10.108.2.91: [               8h] = ipmi_msg_len[ 8b]
10.108.2.91: IPMI Message Header:
10.108.2.91: --------------------
10.108.2.91: [              81h] = rq_addr[ 8b]
10.108.2.91: [               0h] = rq_lun[ 2b]
10.108.2.91: [               7h] = net_fn[ 6b]
10.108.2.91: [              63h] = checksum1[ 8b]
10.108.2.91: [              20h] = rs_addr[ 8b]
10.108.2.91: [               0h] = rs_lun[ 2b]
10.108.2.91: [              3Fh] = rq_seq[ 6b]
10.108.2.91: IPMI Command Data:
10.108.2.91: ------------------
10.108.2.91: [              3Ch] = cmd[ 8b]
10.108.2.91: [               0h] = comp_code[ 8b]
10.108.2.91: IPMI Trailer:
10.108.2.91: --------------
10.108.2.91: [              A8h] = checksum2[ 8b]
```

chu11, 2020-09-22:
hmmm, it seems that one of the SDR records is invalid and its obviously bad enough that I can't figure out the parsing.

Unfortunately the record didn't output b/c it failed before it was output.  Could you try using the `--comma-separated-output` option?  That might get us further, as I think the parsing error occurs when column widths for some pretty formatting is done.

If that doesn't work, I might have to bother you to build a debug build.

`./configure --enable-debug --enable-trace; make; ipmi-sel/ipmi-sel ...`

Thanks

adamwoolhether, 2020-09-22:
```--comma-separated-output``` Didn't work.

I should mention I've been using the version available in standard Centos Repo: ```1.6.1```

I downloaded ```1.6.6```, built a debug build and am able to successfully use ```ipmi-sel```.

1.6.6 seems to fix ```ipmi-sel``` output issue, still unable to  use ```ipmiseld```

chu11, 2020-09-22:
could you try running `ipmi-sel -i`like before, but run it a few times on a problem motherboard, a few minutes apart (to emulate what is done w/ `ipmiseld`).  Want to see if it is changing.

adamwoolhether, 2020-09-22:
Ran it four times, 2-3 minutes apart, on two separate machines. Output is the same.

No activity on the machine, so I initiated a mock-event, shown in the output below.

The daemon now seems to be ok so long as there is the "Number of log entries" is under 909 (with "Free Space" remaining). New logs This raises the question: Is it possible for the daemon to still apply logs above this limit? Or, can i clear the buffer without deleting all the previous logs?

Initial Output:
```
ipmi-sel/ipmi-sel -i -h xxx.13 -u xxx -p xxx
SEL version                            : 1.5
Number of log entries                  : 909
Free space remaining                   : 0 bytes
Recent addition timestamp              : 09/22/2020 - 06:13:47
Recent erase timestamp                 : 07/23/2020 - 05:38:08
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 0
Largest free block                     : 0 allocation units
Maximum record size                    : 1 allocation units
```
After mock event:

```
SEL version                            : 1.5
Number of log entries                  : 909
Free space remaining                   : 0 bytes
Recent addition timestamp              : 09/22/2020 - 06:32:17
Recent erase timestamp                 : 07/23/2020 - 05:38:08
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 0
Largest free block                     : 0 allocation units
Maximum record size                    : 1 allocation units
```

**TEST 2**

Initial Output
```
# ipmi-sel/ipmi-sel -i -h xxx.92 -u xxx -p xxx
SEL version                            : 1.5
Number of log entries                  : 286
Free space remaining                   : 11214 bytes
Recent addition timestamp              : 09/16/2020 - 00:53:05
Recent erase timestamp                 : 05/14/2020 - 00:14:47
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 623
Largest free block                     : 623 allocation units
Maximum record size                    : 1 allocation units
```

After mock event:

```
SEL version                            : 1.5
Number of log entries                  : 288
Free space remaining                   : 11178 bytes
Recent addition timestamp              : 09/22/2020 - 06:06:58
Recent erase timestamp                 : 05/14/2020 - 00:14:47
Get SEL Allocation Information Command : supported
Reserve SEL Command                    : supported
Partial Add SEL Entry Command          : supported
Delete SEL Command                     : supported
Events dropped due to lack of space    : No
Number of possible allocation units    : 909
Allocation unit size                   : 18 bytes
Number of free allocation units        : 621
Largest free block                     : 621 allocation units
Maximum record size                    : 1 allocation units
```

chu11, 2020-09-22:
interesting.  I missed the fact that some of your boards had 0 space, which would of course lead to problems, but unsure
 why there would be issues on the boards w/ space.

> Is it possible for the daemon to still apply logs above this limit? Or, can i clear the buffer without deleting all the previous logs?

I assume you mean "daemon to still read logs above this limit"?  No, if the SEL log is full then the board simply can't store any more logs.  To store more log entries, it will have to be cleared, either manually or via the `ipmiseld` `--clear-threshold` option.

adamwoolhether, 2020-09-23:
> I assume you mean "daemon to still read logs above this limit"? No, if the SEL log is full then the board simply can't store any more logs. To store more log entries, it will have to be cleared, either manually or via the ipmiseld --clear-threshold option.

I suppose my confusion lies here. The BMC's SEL log does continue storing more log entries. This can be verified by logging into the BMC and checking SEL logs, and also with `ipmiseld --test-run`, which will show all additional logs.

If we're able to continue showing additional logs with `--test-run`, can't we force the `ipmiseld` daemon to forward them to rsyslog?

Currently I'm running a cronjob script that runs `ipmiseld --test-run`, compares the output with the current .log file, and appends new entries by piping them to `logger -p local4.info`. It allows our users to retain their historical logs, but It's not the most elegant solution when done at scale for many systems.

Lastly, how can I express my gratitude for your help? Can I get you a coffee or something?

chu11, 2020-09-23:
> I suppose my confusion lies here. The BMC's SEL log does continue storing more log entries. This can be verified by logging into the BMC and checking SEL logs, and also with ipmiseld --test-run, which will show all additional logs.

Huh ... that's interesting.  I'm wondering if the free space / number of entries outputs in the sel info are incorrect, thus leading to the errors we were seeing earlier.

> Lastly, how can I express my gratitude for your help? Can I get you a coffee or something?

haha, thanks.  Supporting FreeIPMI is just part of being part of the open source community.  If you live in the san francisco bay area, perhaps sometime once shelter in place is over.

closed 2020-09-23 by adamwoolhether