chu11/freeipmi-mirror #44: Hardcoded config directory

grawlinson, 2021-05-26:
I forgot to check this particular case before #41 closed.

https://github.com/chu11/freeipmi-mirror/blob/fc0f91d5ebc11b096d06b22f402d51a6bd4d19fd/etc/bmc-watchdog.service.in#L8

I still had `sed -i "s/sysconfig/conf.d/"` in my build scripts. Sorry! That's on me.

Arch Linux has a specific directory for sysconfig `/etc/conf.d/` where other Linux distributions have `/etc/sysconfig` or `/etc/default` or whatever else there is. Sure would be nice if distributions could agree on a few things.

This isn't a major concern because I've already worked around it in the [AUR](https://aur.archlinux.org/packages/freeipmi/) package.

chu11, 2021-05-27:
Doh!  Is there is preferred way to make this configurable/an option?  I haven't been able to find anything semi-universal at the moment.

grawlinson, 2021-05-27:
Could just set a distroconfigdir variable to what it is at the moment (sysconfig) and let distro packagers override the variable at build time.

chu11, 2021-05-27:
My random sampling of other RPMs eventually found `pacemaker` uses:

```
CONFIGDIR=""
AC_ARG_WITH([configdir],
    [AS_HELP_STRING([--with-configdir=DIR],
        [directory for Pacemaker configuration file @<:@SYSCONFDIR/sysconfig@:>@])],
    [ CONFIGDIR="$withval" ]
)
```

which sounds as good as anything else :P  Redhat appears to just hard code /etc/sysconfig in alot of places (based on just random sampling of RPMs).

chu11, 2021-05-28:
@grawlinson  could you give this tree a shot

https://github.com/chu11/freeipmi-mirror/tree/systemconfigdir_option

I ended up naming with `--with-systemconfigdir` ¯\_(ツ)_/¯

chu11, 2021-05-28:
@grawlinson fixed up a few things, can you give the branch a shot, thanks

grawlinson, 2021-05-29:
Works great! I look forward to seeing this in the next release.

I ended up cherry picking & squashing the relevant commits into a single patch that I could apply on top of the 1.6.8 release, as shown in this commit [here](https://aur.archlinux.org/cgit/aur.git/commit/?h=freeipmi&id=7bb0c270846eaaf0a1a6d6ec8aae7a1f5316a11e).

I really appreciate you taking the time to do this, thank you!

closed 2021-05-29 by grawlinson