chu11/freeipmi-mirror #45: HPE DL360 Gen10 servers do not report power supplies properly

sabbene, 2021-07-13:
Hello,

On HPE DL360 Gen10 servers running CentOS7 with free-ipmi version 1.6.8 power supplies are not properly reported.  When I use the HP provided ilorest tool I see the health of the power supplies reported, in this case one is failing, but in the ipmi-sensors output the the status of the power supplies is reported as N/A. 

I have included the output of ipmi-sensors -debug and some sample output below.

[ipmi-sensors-debug.txt](https://github.com/chu11/freeipmi-mirror/files/6811106/ipmi-sensors-debug.txt)

`# dmidecode -s system-product-name
ProLiant DL360 Gen10

# ./ipmi-sensors --version
ipmi-sensors - 1.6.8
Copyright (C) 2003-2015 FreeIPMI Core Team
This program is free software; you may redistribute it under the terms of
the GNU General Public License.  This program has absolutely no warranty.

# ./ipmi-sensors -W discretereading | grep -i power
72 | Power Supply 1   | Power Supply             | N/A        | N/A         | N/A
73 | PS 1 Output      | Power Supply             | N/A        | W           | N/A
74 | Power Supply 2   | Power Supply             | N/A        | N/A         | N/A
75 | PS 2 Output      | Power Supply             | N/A        | W           | N/A
76 | Power Meter      | Other Units Based Sensor | 630.00     | W           | 'OK'
79 | Power Supplies   | Power Supply             | N/A        | N/A         | N/A

# ilorest serverinfo --power
------------------------------------------------
Power Supply 1:
------------------------------------------------
Power Capacity: 800 W
Last Power Output: 629 W
Input Voltage: 201 V
Input Voltage Type: ACHighLine
Hotplug Capable: True
iPDU Capable: False
Health: OK
State: Enabled
------------------------------------------------
Power Supply 2:
------------------------------------------------
Power Capacity: 800 W
Last Power Output: 0 W
Input Voltage: 0 V
Input Voltage Type: Unknown
Hotplug Capable: True
iPDU Capable: False
Health: Critical
State: UnavailableOffline
------------------------------------------------
PowerSupply Redundancy Group 1
------------------------------------------------`

chu11, 2021-07-13:
Could you try the `--bridge-sensors` option (in addition to -W discretereading)?

sabbene, 2021-07-13:
That seems to have done the trick.  Thank you.

```# ipmi-sensors -W discretereading --bridge-sensors | grep -i Power
72 | Power Supply 1   | Power Supply             | N/A        | N/A         | 'Presence detected'
73 | PS 1 Output      | Power Supply             | 340.00     | W           | 'OK'
74 | Power Supply 2   | Power Supply             | N/A        | N/A         | 'Presence detected'
75 | PS 2 Output      | Power Supply             | 275.00     | W           | 'OK'
76 | Power Meter      | Other Units Based Sensor | 625.00     | W           | 'OK'
79 | Power Supplies   | Power Supply             | N/A        | N/A         | 'Fully Redundant'```

closed 2022-02-17 by chu11