chu11/freeipmi-mirror #48: Public key for verify signed tar

Fantu, 2021-11-26:
Hi, I saw that any tar of freeipmi have signature file but from a look to freeipmi site I didn't found the public key to check them.
@chu11 can you tell me where is the public key to check the signature please?

chu11, 2021-11-29:
hi @Fantu, you are the first person to ask in like 11 years :-)  I've uploaded my key to http://keyserver.pgp.com/, thanks!

Fantu, 2021-11-29:
@chu11 thanks, in debian packaging lintian "spot it" debian-watch-does-not-check-gpg-signature [debian/watch],
anyway check of signature is better, for example for see to not download malicious software in case of violation of ftp archives
I'll use add signature check in next version (of debian packages)

Fantu, 2021-12-02:
signature check added for debian packaging and already used and working, thanks
https://salsa.debian.org/debian/freeipmi/-/commit/7175f94c309e07ca6b05a6eb375d64a62c6aba9d

closed 2021-12-02 by Fantu