chu11/freeipmi-mirror #49: Documentation: Add example for conserver

faxm0dem, 2021-12-20:
Hi,

Here's a small addition to the FAQ on how to add an ipmi console to conserver.
Apologies for not succeeding to add this myself to texinfo (/me markdown person)

```
@section How do you setup Conserver with ipmiconsole?

You can find more details in the conserver.cf
(@uref{https://www.conserver.com/docs/conserver.cf.man.html}) manpage.

Here is an example configuration entry from the conserver.cf file:

@example
break 3 { string '"\d\z"'; delay 250; }

default ipmisol {
  break 3;
  motd "Generic IPMI SOL. Use \"Ctrl+E c ?\" for help, \"Ctrl+E c l 0 <key>\" for SysRq, \"Return [ ?\" for ipmi sol help";
  username toor;
  type ipmi;
}

console node42 {
  include ipmisol;
  master conserver.example.com;
  password terces;
  host node42.bmc.example.com;
}
@end example
```

chu11, 2021-12-20:
thanks!  I assume `toor` and `terces` are some random usernames/passwords you selected?  It'd be better to put a "my-username" or "my-password" kinda text.  I can do that in the faq, but just double checking.

faxm0dem, 2021-12-21:
yeah, root and secret in reverse ha ha

chu11, 2022-01-04:
(oops, I never finished merging your change)

Thanks, i'll merge into master and it'll be in the next release of FreeIPMI.

closed 2022-01-04 by chu11