chu11/freeipmi-mirror #51: Static build failure with ipmitool

ffontaine, 2022-01-14:
**Static** build of [ipmitool](https://github.com/ipmitool/ipmitool) with freeipmi raises the following build failure because both packages define `md5_init` and `md5_finish`:

```
/tmp/instance-1/output-1/host/opt/ext-toolchain/bin/../lib/gcc/mips64el-buildroot-linux-uclibc/8.4.0/../../../../mips64el-buildroot-linux-uclibc/bin/ld: /tmp/instance-1/output-1/host/mips64el-buildroot-linux-uclibc/sysroot/usr/lib64/../lib64/libfreeipmi.a(libfreeipmi_la-ipmi-md5.o): in function `md5_init':
/tmp/instance-1/output-1/build/freeipmi-1.6.8/libfreeipmi/libcommon/ipmi-md5.c:130: multiple definition of `md5_init'; ../lib/.libs/libipmitool.a(md5.o):/tmp/instance-1/output-1/build/ipmitool-1.8.18/lib/../src/plugins/lan/md5.c:316: first defined here
/tmp/instance-1/output-1/host/opt/ext-toolchain/bin/../lib/gcc/mips64el-buildroot-linux-uclibc/8.4.0/../../../../mips64el-buildroot-linux-uclibc/bin/ld: /tmp/instance-1/output-1/host/mips64el-buildroot-linux-uclibc/sysroot/usr/lib64/../lib64/libfreeipmi.a(libfreeipmi_la-ipmi-md5.o): in function `md5_finish':
/tmp/instance-1/output-1/build/freeipmi-1.6.8/libfreeipmi/libcommon/ipmi-md5.c:357: multiple definition of `md5_finish'; ../lib/.libs/libipmitool.a(md5.o):/tmp/instance-1/output-1/build/ipmitool-1.8.18/lib/../src/plugins/lan/md5.c:362: first defined here
```

Full build log: http://autobuild.buildroot.org/results/feb/feb68f64d53648f7c95ae825b9c35d086a16154f/build-end.log

I would advise to prefix md5 functions with `freeipmi_` to fix the issue.

chu11, 2022-01-19:
thanks, will fix this up ASAP

chu11, 2022-01-19:
fixed in upstream master, and freeipmi-1-6-0-stable branch

closed 2022-01-19 by chu11