chu11/freeipmi-mirror #53: ipmi-config lan6_conf section disabled

Fantu, 2022-01-25:
Hi, one user on ubuntu reported he use lan6_conf in production and is working
https://bugs.launchpad.net/ubuntu/+source/freeipmi/+bug/1956213
An ubuntu LTS version (Bionic) have it enabled and working, also other minor versions and there was a backport also to another LTS (Xenial), I saw no bugs reported related to it but one about its lack in newer version so is confirmed was used and working.
After he confirmed is working also on unofficial backport of latest version with it enable I'll enable it in next build for debian unstable (and will be also in next ubuntu).
I suppose that can be enabled also upstream, @chu11 what do you think?

chu11, 2022-01-25:
Hi @Fantu, so here's the story, which you can see in commit

```
commit e418a67daf30dd705bea6ffb6cb7e01a48f50278
Author: Albert Chu <chu11@llnl.gov>
Date:   Fri Feb 2 14:08:44 2018 -0800

    Remove IPv6 Configuration Support from 1.6.1 release
    
    Code has not been tested as lost communication with Ubuntu developers
    co-developing with and lack hardware to test on.
    
    Can re-visit when testing has been done.
```

I was working on this was an Ubuntu engineer when the Ubuntu engineer (I think) left Ubuntu and I lost contact.  Since I never had confirmation that the code worked and I couldn't test Lan6_conf on any hardware I had, I was reluctant to release it with FreeIPMI 1.6.1.

I suspect, the code got merged into Ubuntu by someone else?  And I guess it got tested and it works given the discussion in the issue you list above.

If you can just verify that the Lan6_conf code that I have in upstream is basically whats in Ubuntu 18.04, but not in 20.04, I can accept the user's results in that issue as confirmation the code works and we can get it back into FreeIPMI's stable release.  

Since I just did a release, I'd like to wait atleast a little bit (like a month) to do another release if that's ok.

chu11, 2022-01-25:
@fantu ahh i guess earlier versions I iterated with the Ubuntu engineer was what eventually got into 18.04.

If you and your users can confirm the current upstream code works, just let me know and we can get it into the next stable release of FreeIPMI.  If you want, you can just submit the removal of the `#if 0` in a PR to indicate that testing is complete.

wido, 2022-01-26:
@Fantu Thank you for opening the issue. Our organization depends on ipmi-config returning the IPv6 address of the IPMI in our devices and for us this came as a surprise when upgrading to 20.04

We use Dell R6515, Dell R6525 and SuperMicro AS-1123US-TN10RT systems where it all works under Ubuntu 18.04 and is broken under 20.04.

If needed I can provide remote root SSH access to a testing server to verify the code on. Would that work? I would like to have this resolved and backported to Ubuntu 20.04 as this is a regression imho.

We should also make sure it's fixed in Ubuntu 22.04

Fantu, 2022-01-26:
@wido at works I have access to many servers (major dell) and different models, from very old servers with idrac6 up to recent ones with idrac9, the problem are the connections, we are not yet using ipv6 in production because in italy on major of isp there is bad support, for example enabling it on TIM isp will increase much the latency (also in ipv4) :(
I'll do some quick tests on a test server in a few days but if you use it regularly maybe you can try it more (if you want/can) 

about 22.04 is still in development, when I'll upload next debian version enabling lan6_conf will be taken also by ubuntu autosync (up to the feature freeze), for a possible backport request on 20.04 (focal-backports) we will see later; stable proposed update is not possible based on ubuntu 18.04 tests because 20.04 use 1.6.4 that already include significant changes/add to lan6_conf code done upstream and will be not accepted

so for now the important thing is to test lan6_conf as much as possible with the latest version

chu11, 2022-01-26:
@wido would it be possible for you to compile and test the upstream code?  You would have to apply @fantu's patch that he listed above (https://salsa.debian.org/debian/freeipmi/-/blob/master/debian/patches/enable-lan6-conf.patch).

wido, 2022-01-26:
> @wido would it be possible for you to compile and test the upstream code? You would have to apply @Fantu's patch that he listed above (https://salsa.debian.org/debian/freeipmi/-/blob/master/debian/patches/enable-lan6-conf.patch).

Yes, I can. Would that be the main/master branch aka the latest release of freeipmi?

chu11, 2022-01-26:
> Yes, I can. Would that be the main/master branch aka the latest release of freeipmi?

Yes, try the main/master.  Then I would backport the IPv6 patch into the stable 1.6.X line.  Thanks

Fantu, 2022-01-26:
Looking ipmi-config/ipmi-config-category-core-lan6-conf-section.c history stable 1.6 branch have this commit missed (present in master):
https://github.com/chu11/freeipmi-mirror/commit/fcc028c94c9fff430688ffb60bbb1e433ae86d89

For fast/easy tests for normal user (build and install from source may require be developer or tester to avoid issue) there is an unofficial backports for ubuntu 20.04 https://launchpad.net/~fantonifabio/+archive/ubuntu/backports based on latest stable version (1.6.9) that can be easy installed/removed

if needed I'll try a fast packaging from git snapshot when I'll have time to make easy/fast also for other users test

wido, 2022-01-26:
> Looking ipmi-config/ipmi-config-category-core-lan6-conf-section.c history stable 1.6 branch have this commit missed (present in master): [fcc028c](https://github.com/chu11/freeipmi-mirror/commit/fcc028c94c9fff430688ffb60bbb1e433ae86d89)
> 
> For fast/easy tests for normal user (build and install from source may require be developer or tester to be able to avoid issue) there is an unofficial backports for ubuntu 20.04 https://launchpad.net/~fantonifabio/+archive/ubuntu/backports based on latest stable version (1.6.9) that can be easy installed/removed
> 
> if needed I'll try a fast packaging from git snapshot when I'll have time to make easy/fast also for other users test

Using your PPA I have tried the backport on a Dell R6515 server and it kind of works:

```
root@hv-138-a14-09:~# ipmi-config --checkout -S Lan6_Conf
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source_0                  Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable_0                  No
	## Give valid IPv6 address
	IPv6_Static_Address_0                         ::
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length_0           64
	## READ-ONLY
	## IPv6_Static_Address_Status_0               Disabled
fiid_obj_get: 'source': field not found
root@hv-138-a14-09:~#
```

Notice this:

```
fiid_obj_get: 'source': field not found
```

It also does not show me the IPv6 address while the iDRAC has one and is working.

chu11, 2022-01-26:
> Looking ipmi-config/ipmi-config-category-core-lan6-conf-section.c history stable 1.6 branch have this commit missed (present in master): fcc028c

There were a large number of fixes due to analysis from `lgtm.com` that I fixed around that time.  Unlikely that we would need to backport that patch, but we'll keep it in mind, just in case.

chu11, 2022-01-26:
@wido

1) in the version from 18.04 that works, were the `Lan6_conf` IPs listed as `IPv6_Static_Address_0` or `IPv6_Static_Address`?

2) can you run `ipmi-config --checkout -S Lan6_Conf --debug --verbose` and attach the output?

chu11, 2022-01-27:
@wido 

I was able to figure out the cause of the `fiid_obj_get: 'source': field not found` error, so no need for one of the debug outputs I asked for in the previous message.  The fix is in this patch, which you can get if you test w/ the `test_ipv6` branch I mentioned before.

https://github.com/chu11/freeipmi-mirror/commit/af7e4bd16b32527f10ea1e2add87fc765bc1016c

Short answer is there was a cut & paste error from "static" addresses code to "dynamic" addresses code.  This code path had clearly never been tested before.  Thanks for finding it.

wido, 2022-01-27:
> @wido
> 
> I was able to figure out the cause of the `fiid_obj_get: 'source': field not found` error, so no need for one of the debug outputs I asked for in the previous message. The fix is in this patch, which you can get if you test w/ the `test_ipv6` branch I mentioned before.
> 
> [af7e4bd](https://github.com/chu11/freeipmi-mirror/commit/af7e4bd16b32527f10ea1e2add87fc765bc1016c)
> 
> Short answer is there was a cut & paste error from "static" addresses code to "dynamic" addresses code. This code path had clearly never been tested before. Thanks for finding it.

Great! Thanks

@Fantu any chance you can easily build this into a package?

Fantu, 2022-01-27:
@wido did a very fast new build on same ppa

I added also https://github.com/chu11/freeipmi-mirror/commit/fcc028c94c9fff430688ffb60bbb1e433ae86d89 in the new build because I not have time now to adapt the new patch to use without it

wido, 2022-01-27:
> @wido did a very fast new build on same ppa
> 
> I added also [fcc028c](https://github.com/chu11/freeipmi-mirror/commit/fcc028c94c9fff430688ffb60bbb1e433ae86d89) in the new build because I not have time now to adapt the new patch to use without it

I just gave it a try and it works:

```
root@hv-138-a14-09:~# ipmi-config --checkout -S Lan6_Conf
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source_0                  Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable_0                  No
	## Give valid IPv6 address
	IPv6_Static_Address_0                         ::
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length_0           64
	## READ-ONLY
	## IPv6_Static_Address_Status_0               Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_0         
	## READ-ONLY
	## IPv6_Dynamic_Address_0                     2a00:f10:14x:xx:d28e:79ff:fec0:68e
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_0       64
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_0              Active
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_1         
	## READ-ONLY
	## IPv6_Dynamic_Address_1                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_1       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_1              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_2         
	## READ-ONLY
	## IPv6_Dynamic_Address_2                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_2       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_2              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_3         
	## READ-ONLY
	## IPv6_Dynamic_Address_3                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_3       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_3              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_4         
	## READ-ONLY
	## IPv6_Dynamic_Address_4                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_4       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_4              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_5         
	## READ-ONLY
	## IPv6_Dynamic_Address_5                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_5       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_5              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_6         
	## READ-ONLY
	## IPv6_Dynamic_Address_6                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_6       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_6              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_7         
	## READ-ONLY
	## IPv6_Dynamic_Address_7                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_7       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_7              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_8         
	## READ-ONLY
	## IPv6_Dynamic_Address_8                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_8       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_8              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_9         
	## READ-ONLY
	## IPv6_Dynamic_Address_9                     ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_9       255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_9              Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_10        
	## READ-ONLY
	## IPv6_Dynamic_Address_10                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_10      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_10             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_11        
	## READ-ONLY
	## IPv6_Dynamic_Address_11                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_11      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_11             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_12        
	## READ-ONLY
	## IPv6_Dynamic_Address_12                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_12      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_12             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_13        
	## READ-ONLY
	## IPv6_Dynamic_Address_13                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_13      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_13             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_14        
	## READ-ONLY
	## IPv6_Dynamic_Address_14                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_14      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_14             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_15        
	## READ-ONLY
	## IPv6_Dynamic_Address_15                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_15      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_15             Disabled
	## Possible values: Yes/No
	IPv6_Static_Router_Address_Enable             No
	## Give valid IPv6 address
	IPv6_Static_Router_Address_1                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_1              00:00:00:00:00:00
root@hv-138-a14-09:~#
```

It returns back all these addresses, is that correct @chu11 ?

chu11, 2022-01-27:
@wido yes, that is correct given what was implemented.  However, outputting all of those addresses does seem annoying.  Did the 18.04 only output 1 dynamic address?

Perhaps only 1 address should be the default and > 1 addresses under verbose mode.  Let me think about this.

wido, 2022-01-27:
> @wido yes, that is correct given what was implemented. However, outputting all of those addresses does seem annoying. Did the 18.04 only output 1 dynamic address?
> 
> Perhaps only 1 address should be the default and > 1 addresses under verbose mode. Let me think about this.

```root@hv-72-5-34:~# ipmi-config --checkout -S Lan6_Conf
#
# Section Lan6_Conf Comments 
#
# ******** EXPERIMENTAL: This entire section is subject to change in freeipmi 
# 1.6, and missing in freeipmi 1.5. USE AT YOUR OWN RISK. ******* 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Addresses" and set the 
# appropriate routing for the machine. 
#
Section Lan6_Conf
	## READ-ONLY: Supports IPv6-only
	## Supports_IPv6_Only                         No
	## READ-ONLY: Supports IPv6 And IPv4 Simultaneously
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Give valid IPv6 address
	IPv6_Static_Addresses                         ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff
	## READ-ONLY: IPv6 dynamic address
	## IPv6_Dynamic_Addresses                     2a0b:8f80:xx:xx:3eec:efff:fe70:940a
EndSection
root@hv-72-5-34:~# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
root@hv-72-5-34:~# dpkg -l|grep freeipmi
ii  freeipmi-common                        1.4.11-1.1ubuntu4.1                             amd64        GNU implementation of the IPMI protocol - common files
ii  freeipmi-tools                         1.4.11-1.1ubuntu4.1                             amd64        GNU implementation of the IPMI protocol - tools
ii  libfreeipmi16                          1.4.11-1.1ubuntu4.1                             amd64        GNU IPMI - libraries
root@hv-72-5-34:~# 
```

This is the output of Ubuntu 18.04. I think we should stick to that.

chu11, 2022-01-27:
> ## Give valid IPv6 address
> IPv6_Static_Addresses                         ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff [...]

Interesting, I guess the prior engineer decided to put all of the IP addresses on one line.  

We'll go with the suggestion I had above, if not in verbose mode I'll only output 1 static / dynamic address.  I'll try to get a patch out sometime tonight so you can try tomorrow.

chu11, 2022-01-28:
@wido @Fantu I've got a new commit in the `test_ipv6` branch: https://github.com/chu11/freeipmi-mirror/commit/05758316da770a27733377d7e75b350a17940ace

Would you mind trying:

A) `ipmi-config --checkout --section Lan6_conf` - this default output should only output 1 static & 1 dynamic address  for potential config
B) `ipmi-config --checkout --section Lan6_conf --verbose` - this should given you the output you had above, all the potential IPv6 addresses possibilities
C) could you try configuring a static IPv6 address?  (i.e. `ipmi-config --commit --filename config.out` after modifying the config.out file to setup IPv6 static address) I don't think we've determined if that works yet.

Fantu, 2022-01-28:
@wido done another fast build adding latest patch, always in https://launchpad.net/~fantonifabio/+archive/ubuntu/backports

wido, 2022-01-28:
```
root@hv-138-a14-09:~# ipmi-config --checkout -S Lan6_Conf
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. Note that multiple IPv6 addresses can be 
# configured, enable verbose output to view them. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source                    Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable                    No
	## Give valid IPv6 address
	IPv6_Static_Address                           ::
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length             64
	## READ-ONLY
	## IPv6_Static_Address_Status                 Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type           
	## READ-ONLY
	## IPv6_Dynamic_Address                       2a00:f10:xx:xx:d28e:79ff:fec0:68e
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length         64
	## READ-ONLY
	## IPv6_Dynamic_Address_Status                Active
	## Possible values: Yes/No
	IPv6_Static_Router_Address_Enable             No
	## Give valid IPv6 address
	IPv6_Static_Router_Address_1                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_1              00:00:00:00:00:00
root@hv-138-a14-09:~# ipmi-config --checkout -S Lan6_Conf --verbose
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. Note that multiple IPv6 addresses can be 
# configured, enable verbose output to view them. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## READ-ONLY
	## Supports_IPv6_Destination_Address_For_Lan_Alert Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source_0                  Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable_0                  No
	## Give valid IPv6 address
	IPv6_Static_Address_0                         ::
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length_0           64
	## READ-ONLY
	## IPv6_Static_Address_Status_0               Disabled
	## READ-ONLY
[... 112 lines cut ...]
	## IPv6_Dynamic_Address_Source_Type_14        
	## READ-ONLY
	## IPv6_Dynamic_Address_14                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_14      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_14             Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type_15        
	## READ-ONLY
	## IPv6_Dynamic_Address_15                    ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length_15      255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status_15             Disabled
	## Possible values: Yes/No
	IPv6_Static_Router_Address_Enable             No
	## Give valid IPv6 address
	IPv6_Static_Router_Address_1                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_1              00:00:00:00:00:00
root@hv-138-a14-09:~#
```

Seems to work now!

chu11, 2022-01-28:
> Although the keyword under Ubuntu 18.04 was IPv6_Dynamic_Addresses and now it's it IPv6_Dynamic_Address.

@Fantu I should also say, if Ubuntu is interested in supporting backwards compatibility to 18.04, they could maintain a patch just for their tree to do so.  I don't think its right to put this upstream b/c it was never released upstream and with other distros.

I could help you develop / review a patch, but I think most of the work has to come from the Ubuntu side since I don't have the code for 18.04 and don't know what the engineer did.

Fantu, 2022-01-28:
the ubuntu 18.04 code is public: https://packages.ubuntu.com/source/bionic-updates/freeipmi http://archive.ubuntu.com/ubuntu/pool/main/f/freeipmi/freeipmi_1.4.11-1.1ubuntu4.1.debian.tar.xz

about compatibility for thing like the one reported I also don't think is good on debian packages if upstream don't want them, I don't know if a ubuntu developer want do a specific ubuntu patch for this but is improbable, on ubuntu 18.04 patch they added:

> ******** EXPERIMENTAL: This entire section is subject to change in freeipmi 
> # 1.6, and missing in freeipmi 1.5. USE AT YOUR OWN RISK. *******

(an exception if there is another software in debian/ubuntu repository that use old output and "broke" without this compatibility, is there something I don't know or all the scripts are custom?)
anyway after I'll upload 1.6.9-2 with lan6_conf enabled and other upstream patches on debian unstable and will be synchronized to ubuntu 22.04 users can try to require to add a compatibility patch (with 18.04 output) on launchpad

for the moment my main goal is to have ipv6 support in ipmi-config in the next debian and ubuntu (and also next upstream for all it would be good) as it is now widely used, see for example: https://www.google.com/intl/en/ipv6/statistics.html
also because major of work was already done and 2 old ubuntu LTS version had an initial version of it

wido, 2022-01-28:
@chu11 I think you are right, under Ubuntu 20.04 (Freeipmi 1.6.1) we don't need to maintain the backwards compat. We can rename that line.

With that I should way the code works. I still have to test the static IPv6 address, have not tried that.

chu11, 2022-01-28:
>  I still have to test the static IPv6 address, have not tried that.

Once you have tried that, PLMK.  If that works, I think we can merge into the master and put this into the stable 1.6 line.

chu11, 2022-01-28:
@wido @Fantu I found a motherboard that supports ipv6 and found some errors.  I've pushed additional fixes to the test_ipv6 branch.  I haven't been able to confirm ipv6 static addressing works, dunno if its my code or the motherboard.

Fantu, 2022-01-29:
@wido (and any other user want do some fast/easy test) done another fast build adding latest patch, always in https://launchpad.net/~fantonifabio/+archive/ubuntu/backports

wido, 2022-02-03:
Thanks! I can confirm it still works:

```root@hv-138-a14-09:~# ipmi-config --checkout -S Lan6_Conf
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. Note that multiple IPv6 addresses can be 
# configured, enable verbose output to view them. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source                    Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable                    No
	## Give valid IPv6 address
	IPv6_Static_Address                           ::
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length             64
	## READ-ONLY
	## IPv6_Static_Address_Status                 Disabled
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type           
	## READ-ONLY
	## IPv6_Dynamic_Address                       2a00:f10:xx:xx:d28e:79ff:fec0:68e
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length         64
	## READ-ONLY
	## IPv6_Dynamic_Address_Status                Active
	## Possible values: Yes/No
	IPv6_Static_Router_Address_Enable             No
	## Give valid IPv6 address
	IPv6_Static_Router_Address_1                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_1              00:00:00:00:00:00
	## Give valid IPv6 prefix length
	IPv6_Static_Router_Prefix_Length_1            64
	## Give valid IPv6 prefix value
	IPv6_Static_Router_Prefix_Value_1             ::
	## Give valid IPv6 address
	IPv6_Static_Router_Address_2                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_2              00:00:00:00:00:00
	## Give valid IPv6 prefix length
	IPv6_Static_Router_Prefix_Length_2            64
	## Give valid IPv6 prefix value
	IPv6_Static_Router_Prefix_Value_2             ::
EndSection
root@hv-138-a14-09:~#
```

I have not tried the static config yet.

Fantu, 2022-02-11:
hi, I not had time to do some ipv6 tests for now, I want to upload the new debian version within a few days before the ubuntu 22.04 feature freeze but I would prefer to have at least one test on static ipv6 too before upload.
can someone with ipv6 connection and ready-made environment try it please? otherwise I will try to find the time myself in the next days

Fantu, 2022-02-14:
I did a very fast test on old Dell PowerEdge T320 with idrac7 enterprise, I setted a casual ipv6 static from drac web console and query it (I not enabled a real ipv6 connection on the network):
```
ipmi-config --checkout --section=lan6_conf
#
# Section Lan6_Conf Comments 
#
# In the Lan6_Conf section, typical networking configuration is setup. Most 
# users will choose to set an address in "IPv6_Static_Address" and set the 
# appropriate routing for the machine. Note that multiple IPv6 addresses can be 
# configured, enable verbose output to view them. 
#
Section Lan6_Conf
	## READ-ONLY
	## Supports_IPv6_Only                         Yes
	## READ-ONLY
	## Supports_IPv6_And_IPv4_Simultaneously      Yes
	## Possible values: IPv4-Only/IPv6-Only/IPv4-and-IPv6
	IPv6_IPv4_Addressing_Enables                  IPv6_and_IPv4
	## Possible values: Static
	IPv6_Static_Address_Source                    Static
	## Possible values: Yes/No
	IPv6_Static_Address_Enable                    Yes
	## Give valid IPv6 address
	IPv6_Static_Address                           2001:db8:85a3::8a2e:370:7334
	## Give valid prefix length
	IPv6_Static_Address_Prefix_Length             64
	## READ-ONLY
	## IPv6_Static_Address_Status                 Active
	## READ-ONLY
	## IPv6_Dynamic_Address_Source_Type           
	## READ-ONLY
	## IPv6_Dynamic_Address                       ::
	## READ-ONLY
	## IPv6_Dynamic_Address_Prefix_Length         255
	## READ-ONLY
	## IPv6_Dynamic_Address_Status                Disabled
	## Possible values: Yes/No
	IPv6_Static_Router_Address_Enable             No
	## Give valid IPv6 address
	IPv6_Static_Router_Address_1                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_1              00:00:00:00:00:00
	## Give valid IPv6 prefix length
	IPv6_Static_Router_Prefix_Length_1            0
	## Give valid IPv6 prefix value
	IPv6_Static_Router_Prefix_Value_1             ::
	## Give valid IPv6 address
	IPv6_Static_Router_Address_2                  ::
	## Give valid IPv6 mac address
	IPv6_Static_Router_Mac_Address_2              00:00:00:00:00:00
	## Give valid IPv6 prefix length
	IPv6_Static_Router_Prefix_Length_2            0
	## Give valid IPv6 prefix value
	IPv6_Static_Router_Prefix_Value_2             ::
EndSection
```

After I tried an ip change but failed:
```
ipmi-config --checkout --section=lan6_conf >test.out

vi test.out 

ipmi-config --diff --section=lan6_conf --filename=test.out
Lan6_Conf:IPv6_Static_Address - input=`2001:db8:85a3::8a2e:371:7335':actual=`2001:db8:85a3::8a2e:370:7334'

ipmi-config --commit --section=lan6_conf --filename=test.out
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address_Enable'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address_Prefix_Length'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Router_Address_Enable'

ipmi-config --commit --filename=test.out
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address_Source'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address_Enable'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Address_Prefix_Length'
ERROR: Failed to commit `Lan6_Conf:IPv6_Static_Router_Address_Enable'
```
@chu11 Is failed because check if ipv6 is working (and in this case can't) or something else?

chu11, 2022-02-14:
@fantu i'm not sure.  Can you add --debug output?  to limit the output  just do

`ipmi-config --commit --key-pair=lan6_conf:IPv6_Static_Addres=<IP6ADDRESS> --debug`

(hopefully typed that correctly)

chu11, 2022-02-14:
> edit: after I did "ipmi-config --checkout --section=lan6_conf" and saw the ip was changed

wait a second, so did `ipmi-config --commit --key-pair=lan6_conf:IPv6_Static_Address='2001:db8:85a3::8a2e:371:7335' --debug` work, but the commit of the whole file failed?

Fantu, 2022-02-14:
yes, it seems, did another test on whole file with debug if can be useful:
https://paste.debian.net/1230895/

chu11, 2022-02-14:
> ipmi_cmd_get_lan_configuration_parameters_ipv6_static_ip_address: BMC busy

Ahh you are getting BMC Busy errors.  Try the `slowcommit` or `veryslowcommit` workarounds and see if that helps.

If that doesn't work, maybe we should experiment with a hack to try and slow it down even more.

Fantu, 2022-02-14:
@chu11 thanks, with slowcommit still failed but with veryslowcommit worked.
I suppose lan6_conf itself seems ok from some tests done for now, or I'm wrong? (to know if is ok upload next build with lan6_conf to debian unstable or still wait)

chu11, 2022-02-14:
>  I suppose lan6_conf itself seems ok from some tests done for now, or I'm wrong? (to know if is ok upload next build with lan6_conf to debian unstable or still wait)

You said above you didn't try a real ipv6 connection on the network.  So I'd still like to hear from @wido that the static configuration works for him and works with freeipmi tools (ipmi-sensors, etc.) communication before I merge / do a release.

wido, 2022-02-17:
@chu11 Sorry for my late response. I checked and the static config seems to work in my case!

chu11, 2022-02-17:
@wido awesome.  I will merge into master and the stable 1.6 branch.  @Fantu is a new release of FreeIPMI urgent?

Fantu, 2022-02-17:
@wido thanks for testing and report
@chu11 is not urgent, I already uploaded a new build on debian unstable that include all the patches (https://tracker.debian.org/news/1303798/accepted-freeipmi-169-2-source-into-unstable/) and is also taken by ubuntu for next version (is on proposed updates for now)

chu11, 2022-02-17:
@Fantu ok, sounds good.  In that case, I will consider this closed.  @wido thanks for the testing.  We can finally release the code I wrote 4 years ago :-)

closed 2022-02-17 by chu11