chu11/freeipmi-mirror #64: Enable environment variable for password

Frank-Steiner, 2023-05-05:
Hi,
this is an enhancement request. I'd prefer using ipmiconsole over ipmitool because sending of sysrq commands works way easier in ipmiconsole.
There is one drawback, though: for scripting one must pass the passwort with -p in cleartext. That does allow anyone to see it with ps or pgrep because the password isn't masked in the process list.
ipmitool has a nice feature allowing the IPMI_PASSWORD variable to be set and let ipmitool take the password from this variable. Thus, it can be hidden from other users when used in scripts.

So I'd like to ask if you could consider adding support for storing the password in an environment variable, too.

cu,
Frank

chu11, 2023-05-05:
Hi, i'm not necessarily against this idea but traditionally in FreeIPMI scripting with password is done via the FreeIPMI config file.  An alternate default config file can be specified through an option on the command line.  Would this work? (can't lookup details right now, can provide details in follow up post if necessary).

Frank-Steiner, 2023-05-05:
Oh, I missed the --config-file option, but yes, that would definitely work fine! Thanks a lot!

closed 2023-05-05 by Frank-Steiner