chu11/freeipmi-mirror #66: Spelling errors spotted by lintian

Fantu, 2023-05-14:
Hi, lintian spotted some spelling errors:
https://salsa.debian.org/debian/freeipmi/-/jobs/4214071
```
I: libfreeipmi17: spelling-error-in-binary Absolut Absolute [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Accet Accept [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary CENARIO SCENARIO [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Comit Commit [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Communcation Communication [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary DEVELOPPEMENT DEVELOPMENT [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Developement Development [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Develpment Development [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Embeded Embedded [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary ExPet Expect [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary FUNCION FUNCTION [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Folse False [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Gloabl Global [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Goverment Government [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Grammer Grammar [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary INFOMATION INFORMATION [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary InCASE In case [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Informations Information [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Internation International [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Mangement Management [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Millenium Millennium [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Nam Name [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Netowrks Networks [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Neworks Networks [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Ontain Contain [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Optio Option [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Oriente Oriented [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary PRESENSE PRESENCE [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Performace Performance [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Prefered Preferred [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Ressources Resources [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Skelton Skeleton [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Softwares Software [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Sofware Software [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Structual Structural [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Sytems Systems [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Taylored Tailored [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Transmittion Transmission [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Transparity Transparency [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary Updata Update [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary VERION VERSION [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary comunication communication [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary formely formerly [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary limted limited [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
I: libfreeipmi17: spelling-error-in-binary serie series [usr/lib/x86_64-linux-gnu/libfreeipmi.so.17.2.9]
```
Some probably must remain as they are because they are part of code or specifications while some may be correctable errors, I have posted them in case someone want to check and correct where needed and possible

chu11, 2023-05-14:
@fantu thanks for the report.  I'll have to check the specifications to see if they are typos from me or from the specification.  I recall several definitely came from the specification, but I certainly could have introduced some.  At the barest minimum I should add comments, documenting where the spelling errors came from OR if we'll leave them there to avoid programming breakage.

chu11, 2023-05-15:
I just did a check on all of the errors you listed above, and AFAICT (which maybe I missed one), every one of these typos comes from the IANA database of manufacturers http://www.iana.org/assignments/enterprise-numbers.  In some cases I think the typo is in their database, in other cases the spelling is probably correct (e.g. someone's name) or an alternate spelling in non-English native countries (Interestingly, "Nam" appears to be a spacing typo from "Viet Nam").

I'm not sure there's anything I can do about this, as I'm going off what the database says.  They may have corrected some since the last time I updated it.

Is there any type of Lintian "ignore this file for spelling errors" that could be added?  I'd be happy to add the config or comments into the repo.

Fantu, 2023-09-29:
for check spelling error I saw for example codespell as useful, for example with:
`codespell -q 3 -S "./libfreeipmi/spec/ipmi-iana-enterprise-numbers-spec.c" -L "objext"`
-S for exclude specific file that should not be checked like ipmi-iana-enterprise-numbers-spec.c and -L is needed to exclude specific words (in the example I added "objext")

Fantu, 2024-03-29:
I did a PR with codespell config file: https://github.com/chu11/freeipmi-mirror/pull/72
In other projects I added github action instead but as this only a mirror probably is not good
I don't have enough time to fix spelling errors, but I hope this can give a small help to someone want invest time in it