chu11/freeipmi-mirror #67: sensor_units_rate '0x4' not supported

ghost58400, 2023-06-02:
Hello,

I have a problem with reading the state of the FAN sensors of my server, the library gives the following message trying to read the unit rate : 
(ipmi_monitoring_sensor_reading.c, _get_sensor_units, 426): sensor_units_rate '0x4' not supported

Although the reading works in the "ipmimonitoring" command.

[image]

I think the problem is just adding the 0x4 as the right unit in ipmi_monitoring_sensor_reading.c
Here is the full sensor record : 

=====================================================
SDR Full Sensor Record 
=====================================================
[              14h] = record_id[16b]
[               1h] = sdr_version_major[ 4b]
[               5h] = sdr_version_minor[ 4b]
[               1h] = record_type[ 8b]
[              3Bh] = record_length[ 8b]
[               0h] = sensor_owner_id.type[ 1b]
[              10h] = sensor_owner_id[ 7b]
[               0h] = sensor_owner_lun[ 2b]
[               0h] = sensor_owner_lun.reserved[ 2b]
[               0h] = channel_number[ 4b]
[              15h] = sensor_number[ 8b]
[               Ah] = entity_id[ 8b]
[               4h] = entity_instance[ 7b]
[               0h] = entity_instance.type[ 1b]
[               1h] = sensor_initialization.sensor_scanning[ 1b]
[               1h] = sensor_initialization.event_generation[ 1b]
[               0h] = sensor_initialization.init_sensor_type[ 1b]
[               1h] = sensor_initialization.init_hysteresis[ 1b]
[               1h] = sensor_initialization.init_thresholds[ 1b]
[               1h] = sensor_initialization.init_events[ 1b]
[               0h] = sensor_initialization.init_scanning[ 1b]
[               0h] = sensor_initialization.reserved[ 1b]
[               0h] = sensor_capabilities.event_message_control_support[ 2b]
[               1h] = sensor_capabilities.threshold_access_support[ 2b]
[               1h] = sensor_capabilities.hysteresis_support[ 2b]
[               1h] = sensor_capabilities.auto_re_arm_support[ 1b]
[               1h] = sensor_capabilities.entity_ignore_support[ 1b]
[               4h] = sensor_type[ 8b]
[               1h] = event_reading_type_code[ 8b]
[               0h] = threshold_assertion_event_mask.lower_non_critical_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.lower_non_critical_going_high_supported[ 1b]
[               0h] = threshold_assertion_event_mask.lower_critical_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.lower_critical_going_high_supported[ 1b]
[               0h] = threshold_assertion_event_mask.lower_non_recoverable_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.lower_non_recoverable_going_high_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_non_critical_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_non_critical_going_high_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_critical_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_critical_going_high_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_non_recoverable_going_low_supported[ 1b]
[               0h] = threshold_assertion_event_mask.upper_non_recoverable_going_high_supported[ 1b]
[               0h] = lower_threshold_reading_mask.lower_non_critical_threshold_is_comparison[ 1b]
[               1h] = lower_threshold_reading_mask.lower_critical_threshold_is_comparison[ 1b]
[               0h] = lower_threshold_reading_mask.lower_non_recoverable_is_comparison[ 1b]
[               0h] = lower_threshold_reading_mask.reserved[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_non_critical_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_non_critical_going_high_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_critical_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_critical_going_high_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_non_recoverable_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.lower_non_recoverable_going_high_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_non_critical_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_non_critical_going_high_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_critical_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_critical_going_high_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_non_recoverable_going_low_supported[ 1b]
[               0h] = threshold_deassertion_event_mask.upper_non_recoverable_going_high_supported[ 1b]
[               0h] = upper_threshold_reading_mask.upper_non_critical_threshold_is_comparison[ 1b]
[               0h] = upper_threshold_reading_mask.upper_critical_threshold_is_comparison[ 1b]
[               0h] = upper_threshold_reading_mask.upper_non_recoverable_is_comparison[ 1b]
[               0h] = upper_threshold_reading_mask.reserved[ 1b]
[               0h] = readable_threshold_mask.lower_non_critical_threshold_is_readable[ 1b]
[               1h] = readable_threshold_mask.lower_critical_threshold_is_readable[ 1b]
[               0h] = readable_threshold_mask.lower_non_recoverable_threshold_is_readable[ 1b]
[               0h] = readable_threshold_mask.upper_non_critical_threshold_is_readable[ 1b]
[               0h] = readable_threshold_mask.upper_critical_threshold_is_readable[ 1b]
[               0h] = readable_threshold_mask.upper_non_recoverable_threshold_is_readable[ 1b]
[               0h] = readable_threshold_mask.reserved1[ 1b]
[               0h] = readable_threshold_mask.reserved2[ 1b]
[               0h] = settable_threshold_mask.lower_non_critical_threshold_is_settable[ 1b]
[               1h] = settable_threshold_mask.lower_critical_threshold_is_settable[ 1b]
[               0h] = settable_threshold_mask.lower_non_recoverable_threshold_is_settable[ 1b]
[               0h] = settable_threshold_mask.upper_non_critical_threshold_is_settable[ 1b]
[               0h] = settable_threshold_mask.upper_critical_threshold_is_settable[ 1b]
[               0h] = settable_threshold_mask.upper_non_recoverable_threshold_is_settable[ 1b]
[               0h] = settable_threshold_mask.reserved1[ 1b]
[               0h] = settable_threshold_mask.reserved2[ 1b]
[               0h] = sensor_unit1.percentage[ 1b]
[               0h] = sensor_unit1.modifier_unit[ 2b]
[               4h] = sensor_unit1.rate_unit[ 3b]
[               0h] = sensor_unit1.analog_data_format[ 2b]
[              12h] = sensor_unit2.base_unit[ 8b]
[               0h] = sensor_unit3.modifier_unit[ 8b]
[               0h] = linearization[ 7b]
[               0h] = linearization.reserved[ 1b]
[              50h] = m_ls[ 8b]
[               1h] = tolerance[ 6b]
[               0h] = m_ms[ 2b]
[               0h] = b_ls[ 8b]
[               1h] = accuracy_ls[ 6b]
[               0h] = b_ms[ 2b]
[               0h] = sensor_direction[ 2b]
[               0h] = accuracy_exp[ 2b]
[               0h] = accuracy_ms[ 4b]
[               0h] = b_exponent[ 4b]
[               0h] = r_exponent[ 4b]
[               0h] = analog_characteristics_flag.nominal_reading[ 1b]
[               0h] = analog_characteristics_flag.normal_max[ 1b]
[               0h] = analog_characteristics_flag.normal_min[ 1b]
[               0h] = analog_characteristics_flag.reserved[ 5b]
[               0h] = nominal_reading[ 8b]
[               0h] = normal_maximum[ 8b]
[               0h] = normal_minimum[ 8b]
[              FFh] = sensor_maximum_reading[ 8b]
[               0h] = sensor_minimum_reading[ 8b]
[              FFh] = upper_non_recoverable_threshold[ 8b]
[              FFh] = upper_critical_threshold[ 8b]
[              FFh] = upper_non_critical_threshold[ 8b]
[               0h] = lower_non_recoverable_threshold[ 8b]
[               5h] = lower_critical_threshold[ 8b]
[               0h] = lower_non_critical_threshold[ 8b]
[               0h] = positive_going_threshold_hysteresis[ 8b]
[               3h] = negative_going_threshold_hysteresis[ 8b]
[               0h] = reserved[16b]
[              80h] = oem[ 8b]
[              C8h] = id_string_type_length_code[ 8b]
[  BYTE ARRAY ... ] = id_string[16B]
[ 46h 41h 4Eh 20h 50h 53h 55h 31h ]
[ 00h 00h 00h 00h 00h 00h 00h 00h ]
(ipmi_monitoring_sensor_reading.c, _get_sensor_units, 426): sensor_units_rate '0x4' not supported

chu11, 2023-06-02:
hi, i had to investigate this a bit and eventually realized I have this special case handling in `ipmi-sensors` (`ipmimonitoring` is just a script w/ specific options to `ipmi-sensors`).

```
  if (sensor_units_rate != IPMI_SENSOR_RATE_UNIT_NONE)                                                                                      
    {                                                                                                                                       
      /* Special case, RPM is inheritly per minute                                                                                          
       *                                                                                                                                    
       * If vendor specifies a rate other than "per minute", that's                                                                         
       * probably a bug in their SDR.                                                                                                       
       */                                                                                                                                   
      if (sensor_base_unit_type == IPMI_SENSOR_UNIT_RPM                                                                                     
          && sensor_units_rate == IPMI_SENSOR_RATE_UNIT_PER_MINUTE)                                                                         
        rv = snprintf (buf + offset,                                                                                                        
                       buflen,                                                                                                              
                       "%s",                                                                                                                
                       sensor_units[sensor_base_unit_type]);                                                                                
      else                                                                                                                                  
        rv = snprintf (buf + offset,                                                                                                        
                       buflen,                                                                                                              
                       "%s %s",                                                                                                             
                       sensor_units[sensor_base_unit_type],                                                                                 
                       ipmi_sensor_rate_units[sensor_units_rate]);                                                                          
      return (rv);                                                                                                                          
    }                                                                          
```

(an aside, I notice there's a typo above, that should be "inherently")

So perhaps need to add this special case handling to `libipmimonitoring`.

chu11, 2023-06-02:
If possible, could you apply this patch and try it out?

```C
diff --git a/libipmimonitoring/ipmi_monitoring_sensor_reading.c b/libipmimonitoring/ipmi_monitoring_sensor_reading.c
index 3499161..f7a4b32 100644
--- a/libipmimonitoring/ipmi_monitoring_sensor_reading.c
+++ b/libipmimonitoring/ipmi_monitoring_sensor_reading.c
@@ -423,8 +423,12 @@ _get_sensor_units (ipmi_monitoring_ctx_t c,
 
   if (sensor_units_rate != IPMI_SENSOR_RATE_UNIT_NONE)
     {
-      IPMI_MONITORING_DEBUG (("sensor_units_rate '0x%X' not supported", sensor_units_rate));
-      return (IPMI_MONITORING_SENSOR_UNITS_UNKNOWN);
+      /* Special case, "RPM" is inherently "per minute" */
+      if (!(sensor_base_unit_type == IPMI_SENSOR_UNIT_RPM
+            && sensor_units_rate == IPMI_SENSOR_RATE_UNIT_PER_MINUTE)) {
+        IPMI_MONITORING_DEBUG (("sensor_units_rate '0x%X' not supported", sensor_units_rate));
+        return (IPMI_MONITORING_SENSOR_UNITS_UNKNOWN);
+      }
     }
```

chu11, 2023-06-06:
I'm confident in this fix, so I'm going to go ahead and merge it.  I'd like to do a release half soon, so would be good to know if it works.  I've already pushed it into the stable branch here:

https://github.com/chu11/freeipmi-mirror/tree/freeipmi-1-6-0-stable

ghost58400, 2023-06-06:
Hello,

Sorry for the late reply, the way I use libipmimonitoring is through the IPMI plugin of [Netdata](https://github.com/netdata/netdata).

Unfortunately, I will not be able to test this quickly because the problem appears on a production server, on which I cannot install manually compiled packages...

I think that the issue you found looks legit and you can include this patch in the next release without problem.

ghost58400, 2023-08-15:
Hello,

I am back with good news, I installed freeipmi 1.6.11 and the fix works perfectly !
I close the issue, thanks for the work !

closed 2023-08-15 by ghost58400