chu11/freeipmi-mirror #71: Fails to build for implicit-function-declaration now error by default in Debian (and derivates) from dpkg 1.22.6

Fantu, 2024-03-13:
Freeipmi fails to build for implicit-function-declaration now error by default in Debian (and derivates) from dpkg 1.22.6 to avoid an issue, more details here: https://wiki.debian.org/qa.debian.org/FTBFS#A2024-03-13_-Werror.3Dimplicit-function-declaration

Was reported in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1066711

Relevant part (hopefully):
> /bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../config  -I../common/debugutil -I../common/miscutil -I../common/portability -I../libfreeipmi/include -I../libfreeipmi/include -I../libfreeipmi -DIPMI_IPCKEY=\"/var/lib/freeipmi/ipckey\" -DIPMI_DEBUG_IPCKEY=\"../libfreeipmi/driver/ipmi-semaphores.h\" -D_REENTRANT -Wdate-time -D_FORTIFY_SOURCE=2  -g -O2 -Werror=implicit-function-declaration -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -c -o spec/libfreeipmi_la-ipmi-entity-ids-spec.lo `test -f 'spec/ipmi-entity-ids-spec.c' || echo './'`spec/ipmi-entity-ids-spec.c
> sel/ipmi-sel-string-supermicro-x10dimm-common.c: In function ‘sel_string_output_supermicro_x10dimm_event_data2_event_data3’:
> sel/ipmi-sel-string-supermicro-x10dimm-common.c:87:14: error: implicit declaration of function ‘sel_string_output_supermicro_dimm_event_data2_event_data3’; did you mean ‘sel_string_output_supermicro_x10dimm_event_data2_event_data3’? [-Werror=implicit-function-declaration]
>    87 |   if ((ret = sel_string_output_supermicro_dimm_event_data2_event_data3 (ctx,
>       |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |              sel_string_output_supermicro_x10dimm_event_data2_event_data3
> /bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../config  -I../common/debugutil -I../common/miscutil -I../common/portability -I../libfreeipmi/include -I../libfreeipmi/include -I../libfreeipmi -DIPMI_IPCKEY=\"/var/lib/freeipmi/ipckey\" -DIPMI_DEBUG_IPCKEY=\"../libfreeipmi/driver/ipmi-semaphores.h\" -D_REENTRANT -Wdate-time -D_FORTIFY_SOURCE=2  -g -O2 -Werror=implicit-function-declaration -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -c -o spec/libfreeipmi_la-ipmi-fru-chassis-types-spec.lo `test -f 'spec/ipmi-fru-chassis-types-spec.c' || echo './'`spec/ipmi-fru-chassis-types-spec.c
> cc1: some warnings being treated as errors
> make[3]: *** [Makefile:2703: sel/libfreeipmi_la-ipmi-sel-string-supermicro-x10dimm-common.lo] Error 1

The full build log is available from:
http://qa-logs.debian.net/2024/03/13/freeipmi_1.6.13-2_unstable.log

chu11, 2024-03-23:
Thanks.  I think the function prototype was simply missing from an internal header file.   Should be fixed in the next release.

closed 2024-03-23 by chu11

Fantu, 2024-03-24:
@chu11 thanks, unfortunately after applying the patch I saw that it fails again due to another warning.
https://salsa.debian.org/debian/freeipmi/-/jobs/5489636
Sorry, I didn't think to look at the previous build's warnings successfully to see if there were others.
According to this, it should seem that there are no others besides those 2:
https://qa.debian.org/bls/packages/f/freeipmi.html

chu11, 2024-03-24:
@Fantu thanks for the report.  I think there was a mistake in the header guard definitions.  Pushed a fix to the stable branch (not in master, was previously fixed there)

c3d1a3f5c550940577076ce2c4595b470ae963c3

Fantu, 2024-03-25:
@chu11 I applied also https://github.com/chu11/freeipmi-mirror/commit/c3d1a3f5c550940577076ce2c4595b470ae963c3 but still fails:
https://salsa.debian.org/debian/freeipmi/-/jobs/5496348

I tried a fast look, `ipmi_sensors_oem_intel_s2600wp_output_oem_record` is defined in ipmi-sensors-oem-intel-s2600wp.h, but I don't find it in include list in start of ipmi-sensors-oem-intel.c

chu11, 2024-03-25:
crud, sorry about that.  I went between master & stable branch and didn't see that was missing too.

6ac6f816dbfc83d8cd4f4f163b0f8ff1062d883e