chu11/freeipmi-mirror #76: 机翻 · 通用领域 Can ipmi-sensors access sensors with sensor num exceeding 255?

linshuhua-linshuhua, 2025-08-10:
机翻 · 通用领域
Can ipmi-sensors access sensors with num greater than 255 by setting lun to 1 like ipmitool

chu11, 2025-08-12:
By IPMI definition the sensor number is 1 byte, so it can only have a maximum of 255.

I think what you're asking is that by utilizing other external factors, such as the LUN, can you access more than 255?  The answer is yes.

chu11, 2025-09-02:
I don't understand your question.

Sensor numbers cannot exceed 255, it is impossible.

Are you asking how to specify sensors ont he command line?  You can pass record ids on the command line with --record-ids.

There is no known issue regarding sensor numbers.

linshuhua-linshuhua, 2025-09-04:
What I mean is, the same machine ipmitool sdr elist can normally obtain sensor data with sensor num exceeding 255. Why is that ipmi-sensors -h ip -u user -p password -D LAN_2_0 --sdr-cache-recreate --quiet-cache list  Can't it be done?